BBN-Q/QGL #4: Compiling >20k long sequence saturates RAM (>12 GB)

dieris, 2016-02-05:
Compiling the following sequences takes ~10 minutes and in Perseus, using ~12 GB RAM and severely slowing it down until `seqs` is deleted. 

```
import csv  
gatedic = {}  
gatedic['1'] = (X90,a)
gatedic['2'] = (X90,q)
gatedic['3'] = (Y90,a)
gatedic['4'] = (Y90,q)
seqs=[]
with open('gst16-2q.csv', 'rb') as csvfile:
    gatemat = csv.reader(csvfile, delimiter=',', quotechar='|')
    for ind,row in enumerate(gatemat):
        seq = []
        for key in row:
            if key=='5':
                seq+= ZX90_CR(a,q)
            else:
                if key!='0':
                    seq+=[gatedic[key][0](gatedic[key][1])]
        seq+=[MEAS(a)*MEAS(q)]
        seqs+=[seq]
    seqs+=create_cal_seqs((a,q),250)

filenames = compile_to_hardware(seqs,'Test\Test')
```

[gst16-2q.zip](https://github.com/BBN-Q/QGL/files/119264/gst16-2q.zip)

caryan, 2016-05-25:
@dieris  is this still a relevant example? I'd like to do some profiling.

dieris, 2016-05-25:
yes, without any Repeat or subroutines, compiling 27k seqs is still prohibitive (PC hangs for >10 minutes)

dieris, 2016-07-14:
Closed by https://github.com/BBN-Q/QGL/pull/60

closed 2016-07-14 by dieris