BBN-Q/QGL #5: Reset sequence improvement

caryan, 2016-02-17:
The `Reset` sequence is a bit cryptic to follow. A few issues

1. docstring update
2. more liberal comments required
3. update for fixed `count` vs `fbcount`
4. don't need `count` anymore

blakejohnson, 2016-02-17:
A couple possible re-writes:

```python
def Reset2(qubits, measDelay = 1e-6, signVec = None, doubleRound = True, buf = 30e-9, showPlot=False, measChans=None, docals=True, calRepeats=2):
	if measChans is None:
		measChans = qubits

	if signVec == None:
		signVec = [0]*len(qubits)

	# for each qubit, build the set of feedback action to perform when
	# receing a zero or one in the comparison register
	FbGates = []
	for ct, q in enumerate(qubits):
		if signVec[ct] == 0:
			FbGates.append([gate(q) for gate in [Id, X]])
		else: # inverted logic
			FbGates.append([gate(q) for gate in [X, Id]])
	FbSeq = [reduce(operator.mul, x) for x in product(*FbGates)]

	# load register
	resetSeq = [Id(qubits[0], measDelay), qwait('CMP'), Id(qubits[0], buf)]
	# create a branch for each possible comparison value
	for ct in range(2**len(qubits)):
		resetSeq += qif(ct, [FbSeq[ct]])
	resetSeq += [MEAS(*measChans)]

	if doubleRound:
		pass # do something

	# the copy isn't quite sufficient because we need to create all new labels
	seqs = [prep + copy(resetSeq) for prep in create_cal_seqs(qubits,1)]

	# add final qwait('CMP')
	for seq in seqs:
		seq.append(qwait('CMP'))

	if docals:
		seqs += create_cal_seqs(qubits, calRepeats, measChans=measChans)

	return seqs
```

or alternatively with a `qfunction`:

```python
@qfunction
def qreset(qubits, signVec, measDelay, buf):
	# for each qubit, build the set of feedback action to perform when
	# receing a zero or one in the comparison register
	FbGates = []
	for ct, q in enumerate(qubits):
		if signVec[ct] == 0:
			FbGates.append([gate(q) for gate in [Id, X]])
		else: # inverted logic
			FbGates.append([gate(q) for gate in [X, Id]])
	FbSeq = [reduce(operator.mul, x) for x in product(*FbGates)]

	# load register
	seq = [Id(qubits[0], measDelay), qwait('CMP'), Id(qubits[0], buf)]
	# create a branch for each possible comparison value
	for ct in range(2**len(qubits)):
		seq += qif(ct, [FbSeq[ct]])
	seq += [MEAS(*qubits)]

	return seq

def Reset3(qubits, measDelay = 1e-6, signVec = None, doubleRound = True, buf = 30e-9, showPlot=False, measChans=None, docals=True, calRepeats=2):
	if measChans is None:
		measChans = qubits

	if signVec == None:
		signVec = [0]*len(qubits)
	signVec = tuple(signVec)

	seqs = [prep + [qreset(qubits, signVec, measDelay, buf)] for prep in create_cal_seqs(qubits,1)]
	if doubleRound:
		for seq in seqs:
			seq.append(qreset(qubits, signVec, measDelay, buf))

	# add final qwait('CMP')
	for seq in seqs:
		seq.append(qwait('CMP'))

	if docals:
		seqs += create_cal_seqs(qubits, calRepeats, measChans=measChans)

	return seqs
```

caryan, 2016-04-01:
Closed by #10

closed 2016-04-01 by caryan