BBN-Q/QGL #35: update waveform library post compiling

caryan, 2016-05-24:
For sequences where are a burden to recompile and we have some small set of cannonical pulses it would be nice to be able to update the waveform library in place after recalibrating pulse parameters. We'd need something like:

1. pickle the offset dictionary with usable keys like "X90(q1)"
2. a helper function to update a h5 file given this dictionary

suggestion from @marcusps and @dieris

caryan, 2016-06-01:
@dieris and @matthewware 

b023a7e4675d58ba1d83630568318c8879e98c06 provides an initial attempt at this.  It looks something like:

```python
In [1]: from QGL import *

In [2]: q1 = QubitFactory("q1")

In [3]: q2 = QubitFactory("q2")

In [4]: seqs = [[X90(q1)] + ZX90_CR(q1, q2) + [MEAS(q1)]]

In [5]: from QGL.drivers import APS2Pattern

In [6]: APS2Pattern.SAVE_WF_OFFSETS = True

In [7]: files = compile_to_hardware(seqs, "Test/Test")
Compiled 1 sequences.

In [8]: q1.pulseParams['piAmp'] = 0.1

In [9]: pulses = [X90(q1), X(q1), Y90(q1), Y(q1), X90(q2), X(q2), Y90(q2), Y(q2), ZX90_CR(q1, q2), MEAS(q1)]

In [10]: PatternUtils.update_wf_library(pulses, "/home/cryan/Programming/Repos/QGL/awg/Test/Test")
Updating pulses for BBNAPS4
	Updating MEAS(M-q1) at offset 0
Offset file not found for BBNAPS2, skipping pulses ['Y(q2)', 'X90(q2)', 'X(q2)', 'Y90(q2)']
Updating pulses for BBNAPS1
	Y(q1) not found in offsets so skipping
	Updating X90(q1) at offset 48
	Updating X(q1) at offset 0
	Y90(q1) not found in offsets so skipping
Updating pulses for BBNAPS3
	Updating echoCR_first_half_top(CR12) at offset 4
	Updating echoCR_first_half_rise(CR12) at offset 100
	Updating echoCR_first_half_fall(CR12) at offset 124
	Updating echoCR_second_half_rise(CR12) at offset 28
	Updating echoCR_second_half_top(CR12) at offset 76
	Updating echoCR_second_half_fall(CR12) at offset 52

In [11]: 
```

dieris, 2016-07-07:
Fixed by https://github.com/BBN-Q/QGL/pull/42 and https://github.com/BBN-Q/QGL/pull/50

closed 2016-07-07 by dieris