BBN-Q/QGL #43: Have `compile_to_hardware` save the "code" it is compiling.

blakejohnson, 2016-06-24:


blakejohnson, 2016-06-24:
So, to do this, we don't even need the `meta` package, we just need to effectively to do:
```python
fid = open("output.txt", "w")
fid.write(str(seqs))
fid.close()
```
This would be a good first step, but it is not exactly human readable. Do we care?

If we do want it to be human readable, then I would propose saving a copy of the channel dictionaries, and then expanding the `Pulse` pretty printers so that they output just enough information to reproduce the object when reconstructed from their `str` representation.

caryan, 2016-06-26:
Since this will mainly be for experiment archaeology or debugging purposes, human readable seem desirable.

PR #49 (2016-07-05): Save input code to file.
Most of the work was in getting `str(Pulse)` to behave. Consequently, we can just use ipython's pretty printer to get reasonable output. I've put this behind an `if not qgl2` guard under the assumption that if you are using QGL2, you already have the code somewhere.

Fixes #43.
closed 2016-07-05 by caryan