BBN-Q/QGL #44: Produce a meta output file with the complete program description

blakejohnson, 2016-06-24:
i.e. we want to have:
- [ ] a dictionary from instruments to their compiled sequence files
- [ ] some equivalent information for receivers/digitizers
    - [ ] including some representation of the difference between a segment and a capture event (i.e. if there are multiple `MEAS`s in a sequence
- [ ] label information for swept parameters

caryan, 2016-06-26:
For the receiver section we also need some way to describe variable number of captures per segment. It might also be the place to think about what an experiment debug record would look like.

caryan, 2016-07-15:
Some initial work in #49 to save the pulse sequence string

blakejohnson, 2017-03-27:
`ReceiverChannel`s add in #91.