BBN-Q/QGL #48: Square pulse identifier

dieris, 2016-07-01:
It seems that [this commit](https://github.com/BBN-Q/QGL/commit/c66a865255f46239f5f5d1fb0a354a7683c72e28) conflicts with this [condition](https://github.com/BBN-Q/QGL/blob/c66a865255f46239f5f5d1fb0a354a7683c72e28/QGL/drivers/APS2Pattern.py#L369) to select the relevant waveform identifier. The length of a square pulse may still be relevant when updating the waveform library in https://github.com/BBN-Q/QGL/commit/c139acc563e55bf450baecc27b45bb50cba7439f

caryan, 2016-07-01:
@dieris can you be more explicit about what fails and what changed for you: it seems nothing should have changed about when `isTimeAmp` is set to `True`.

dieris, 2016-07-01:
in my understanding, the square pulse was not a TA pulse before, now it is. So the waveform misses the length attribute and every square pulse is considered to be the same. As a result, if I try to update the amplitude of a square pulse in a sequence, it can't find the corresponding offset. I can try to make an example where it fails, I don't have it right now

blakejohnson, 2016-07-01:
In terms of the waveform library, when square pulses are represented as time/amp pulses, then they should have identical signatures.

In the scenario you are describing, do you have two square pulses serving different functions that happen to have the same initial amplitude, but you want to update one of them later (and not the other)?

dieris, 2016-07-01:
I had this problem when updating CR pulses, where the flat part has square shape. I'm trying to reproduce the error

dieris, 2016-07-01:
ok, this seems to fail consistently

```
seq = [ZX90_CR(q3,q1)]
from QGL.drivers import APS2Pattern
APS2Pattern.SAVE_WF_OFFSETS = True
compile_to_hardware(seq,'Test\Test')
PatternUtils.update_wf_library([ZX90_CR(q3,q1)], os.path.normpath(os.path.join(seqpath, 'Test/Test')))
```
[cfg.zip](https://github.com/BBN-Q/QGL/files/344168/cfg.zip)

caryan, 2016-07-03:
OK I see what's going on now.  Fix coming.

closed 2016-07-04 by dieris

caryan, 2016-07-05:
Closed by #50