BBN-Q/QGL #77: Non-unique pulse key?

dieris, 2016-09-13:
This doesn't make any sense to me, but dragScaling = -1 and -2 have the same key (though different shape in Q as expected)

```
X(q3, dragScaling = -1).hashshape() == X(q3, dragScaling = -2).hashshape()
True
```

caryan, 2016-09-13:
I'm :confused:...  `Pulse.hashshape()` is just hashing the shapeParams but

```python
In [1]: from QGL import *
In [2]: q1 = QubitFactory("q1")
In [3]: a = X(q1, dragScaling = -1)
In [4]: b = X(q1, dragScaling = -2)
In [5]: for item in frozenset(a.shapeParams.items()):
   ...:     print(item, hash(item))
   ...:     
('shapeFun', <function gaussian at 0x7ff9716c38c8>) -8978732342065354207
('sigma', 5e-09) 6129484651352346146
('length', 2e-08) -4582154280831537373
('piAmp', 1.0) 7177149320524611909
('dragScaling', -1) 8624729461345775280
('cutoff', 2) 6109634325142089946
('pi2Amp', 0.5) 7320135987436524444

In [6]: for item in frozenset(b.shapeParams.items()):
   ...:     print(item, hash(item))
   ...:     
('shapeFun', <function gaussian at 0x7ff9716c38c8>) -8978732342065354207
('sigma', 5e-09) 6129484651352346146
('length', 2e-08) -4582154280831537373
('piAmp', 1.0) 7177149320524611909
('dragScaling', -2) 8624729461345775280
('cutoff', 2) 6109634325142089946
('pi2Amp', 0.5) 7320135987436524444

In [7]: hash(('dragScaling', 1))
Out[7]: -8624729461346662743

In [8]: hash(('dragScaling', 2))
Out[8]: -8624729461347745268

```

caryan, 2016-09-13:
Seems to be a hash collision with hashing a tuple of something and -1 or -2 e.g.

```python
In [1]: [hash(('silly', x)) for x in [-1,-2,3]]
Out[1]: [-3277634604971975924, -3277634604971975924, 3277634604968923411]

In [2]: [hash(('', x)) for x in [-1,-2,3]]
Out[2]: [-3713080549410298119, -3713080549410298119, 3713080549411575706]

In [3]: [hash((1, x)) for x in [-1,-2,3]]
Out[3]: [-3713081631936380644, -3713081631936380644, 3713081631933328131]

```

caryan, 2016-09-13:
Actually it's caused by a hash collision between -1 and -2: both hash to -2 ????

caryan, 2016-09-13:
Seems to be that Python uses -1 to denote a hash failure so -1 gets shifted to -2 so we may want to salt any integers.

dieris, 2016-09-13:
interesting, I guess I'll shy away from those DRAG integers... but it seems dangerous as it could happen elsewhere

blakejohnson, 2016-09-13:
Yeah, definitely dangerous. Colm and I just discussed a proper fix.