BBN-Q/QGL #96: QGL IR

blakejohnson, 2017-03-29:
This issue is to track discussions about supporting a simple, pure text IR for QGL. Some initial notion of what that might look like from this morning's discussion:

```
QBIT_q1 = Qubit('q1')
QBIT_q2 = Qubit('q2')
OPAQUE VOTE

MEAS(QBIT_q1) -> 0x0010
MEAS(QBIT_q2) -> 0x0011
VOTE(0x0010, 0x0011) -> 0x0013
LoadCmp(0x0013)
CmpEq(0)
Goto(BlockLabel('if_1'))
Id(QBIT_q1)
Goto(BlockLabel('if_end_1'))
BlockLabel('if_1')
X(QBIT_q1)
BlockLabel('if_end_1')
X90(QBIT_q1)
```

caryan, 2017-03-29:
The `LoadCmp` seems very APS2 specific and adds an extra instruction. Perhaps the comparison operators (`CmpEq` etc) should take an address and the APS2 would translate that to `LoadCmp(addr)`.

blakejohnson, 2017-03-29:
Sure, I like that.