BBN-Q/QGL #113: make location of 'config.json' configurable

dellard, 2017-07-24:
Reported by Robert McGurrin: 

When QGL is imported, config.json is read from the QGL package's directory (see QGL/config.py).  I'd like to change this so that the config.json file can be read from alternate locations.
 
I'd like to add the capability of loading the config.json file from the user's local directory if it exists: otherwise the default config.json file should be loaded from the QGL/config.py when the package is loaded.

I'd also like to add the capability for the User to specify the config file using  a function like:
config('/somepath/config.json')

The purpose of these changes is to make it easier for the user to change configurations.

dellard, 2017-07-24:
I put together a simple approach to solving this problem, but it doesn't work properly with `unittest` because it only sets the path once (and then it stays set to that value for all subsequent tests).

I would imagine that this sort of issue has come up before, so I'm going to see if I can find a ready-made solution.  Alternatively, we can live without unit tests for this feature.

grahamrow, 2017-07-25:
In terms of hacked-together solutions, we could create a test script for this feature that gets run as a subprocess in the main `unittest` suite.

dellard, 2017-07-25:
Yes, shelling out to a subprocess would do it, I think.

dellard, 2017-08-08:
Resolved by 6aaca5077baecf98c790c4d5629729d07a29c0fb

closed 2017-08-08 by dellard