BBN-Q/QGL #114: Reconsider delay accumulation convention

grahamrow, 2017-07-28:
Right now QGL automatically offsets pulses by the minimum delay to avoid negative delays. This is awkward across multiple APS slices if you want to delay a slave trigger, since it automatically pushes quadrature channel pulses off into the future, thereby canceling the intention. What should we do?

blakejohnson, 2017-08-07:
Yes, I have been bitten by that oddness before with the slave trigger. But, this isn't a simple matter of "avoiding" negative delays. What does it even mean for a master AWG channel to have a negative delay, given that the AWG output is triggered?

gribeill, 2017-08-07:
Possible solutions, off the top of my head:

1. Channel timing is preserved across AWGs, with all delays relative to 0 delay on the master AWG. 
2. Channel timing is preserved across AWGs, with all delays relative to the leftmost channel or marker on the master AWG
3. Channel timing is preserved across AWGs, with all delays relative to the first pulse in the whole sequence.
4. Channel delays only refer to a specific AWG, with negative delays being an error. t=0 for each AWG happens when it receives its trigger.
5. Channel delays only refer to a specific AWG, with negative delays being an error for the master AWG and delays for the slaves relative to the `tx_channels` on that AWG.
6. What we are doing now, where delays move pulses left or right in time.

I think 2, 3, and 5 are clearly bad ideas, since they require the user to infer too much state when setting up delays. 1 is misleading, since it mixes "logical" delays that the user wants between pulses and "physical" delays between pulses due to cabling, etc... I might have missed something but I think (4) would be my preferred solution -- although I suspect this would require some additional cleverness from the QGL compiler...

blakejohnson, 2017-08-08:
It is somewhat of a hack, but I think you would also get sane behavior if the slave trigger channel were really forced to have delay = 0, i.e. the slave trigger output would be invariant under changes to any other channel delay. This would allow, for example, putting negative delays on slave AWGs and have The Right Thing Happen™.

This would also imply that negative delays on the master AWG would be an error.