BBN-Q/QGL #118: Either document how to setup git-lfs, or get rid of it

dellard, 2017-08-08:
The example test output files are represented in the git repo by references to git lfs objects.  It doesn't look like this dependency is mentioned anywhere (or at least not blatantly enough to catch my eye), so my builds were failing.

I don't like adding dependencies (especially ones that aren't compatible with some platforms), and I don't know that we're benefitting from using git-lfs here anyway.  I would recommend either getting rid of it (if that's possible), or at minimum adding documentation to the README.md explaining how to acquire, install, and use git-lfs correctly.

grahamrow, 2017-08-15:
I think documentation is the best short-term solution, but as long as our `git clone`s don't become enormous we should consider alternatives.

blakejohnson, 2017-08-15:
We are using Git LFS because it solves the problem that standard git is not good at tracking non-text files. Our tests depend on many 10s of MB of HDF5 files, which have changed several times over time. Using Git LFS keeps the primary git repo down to a manageable size.