BBN-Q/QGL #121: Other trigger pulses.

gribeill, 2017-08-14:
Would be nice to be able to specify trigger pulses that are not necessarily associated with a qubit measure or control channel.

grahamrow, 2017-08-14:
Hmm, that's an interesting idea. We could feasible add a `m = MarkerFactory("BBNAPS2-12m4")` and then create a pulse primitive like `TRIG(M)`. Or we could just call a trigger directly on a channel string.

gribeill, 2017-08-15:
Great minds think alike... check out #122

grahamrow, 2017-08-15:
Is there anything un-QGLlike about this proposition? We've been striving to make the YAML more qubit centric than channel centric. This opens the doors for some weird looking sequences whose functions are somewhat obscured by lacking the physical connectivity of the experiment. Of course a gate language that's agnostic to experimental topology is either impossible to write or useless, so maybe this is completely fine?

gribeill, 2017-08-15:
I toyed with having `markers` be another sub-category of `qubits` like `control` or `measure` but that didn't seem right either.

I agree in principle that this somewhat breaks the YAML model but I think it's unavoidable on some level as you point out. We already have special-case handling for digitizer triggers and the slave trigger baked into the channel library so this is already happening under the hood. I think it's fine to have triggers be another category of events that QGL can generate along with qubit pulses and edge pulses. I'm open to better ideas though...

matthewware, 2017-08-15:
We could think about extending the idea of a trigger/gate on a measurement channel to bias amps say...  Though this might not be the best abstraction.

dieris, 2017-08-15:
@gribeill It seems like a good feature to have, but can you give us a use example so we can see how general this needs to be? I encountered a similar need before, but it was either for:
1) Triggering a second digitizer (different from the one associated with the receiver). In this case I'd use a `MEAS(v)`, with `v` a virtual qubit.
2) Gating pulses, which is already supported in QGL.

gribeill, 2017-08-15:
So to respond to @matthewware / @dieris : 

Gate channels are still there (and now work with the yaml config!) -- defined by a `gate:` key under `measure:` or `control:`, but don't quite cover my need... I need to trigger a parametric amplifier pump well before the measurement pulse is turned on, and have the "trigger" be much longer than the measurement pulse.

In the JSON days I would do this using the virtual qubit approach but this doesn't really work anymore -- the YAML demands a complete qubit specification for each qubit, real or not (we could of course change this!) -- but this seems like a bad and cumbersome way to just get a digital marker output...

gribeill, 2017-08-15:
Closed by #122

closed 2017-08-15 by gribeill