BBN-Q/QGL #135: error in unittest: test_RB_SimultaneousRB_AC (APS2)

dellard, 2017-11-15:
Observed on the HEAD of master, with the default config.json file.

(it's possible my other config files are out of sync, but this looks like a code problem, not a config problem)

```
ERROR: test_RB_SimultaneousRB_AC (tests.test_Sequences.TestAPS2)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/dellard/RF/QGL/tests/test_Sequences.py", line 374, in test_RB_SimultaneousRB_AC
    SimultaneousRB_AC((self.q1, self.q2), (seqs1, seqs2))
  File "/home/dellard/RF/QGL/QGL/BasicSequences/RB.py", line 294, in SimultaneousRB_AC
    metafile = compile_to_hardware(seqsBis, 'RB/RB')
  File "/home/dellard/RF/QGL/QGL/Compiler.py", line 349, in compile_to_hardware
    wireSeqs = compile_sequences(seqs, channels)
  File "/home/dellard/RF/QGL/QGL/Compiler.py", line 486, in compile_sequences
    wires = compile_sequence(seqs[0], channels)
  File "/home/dellard/RF/QGL/QGL/Compiler.py", line 551, in compile_sequence
    wires, chan, block.pulses[chan].frameChange)
  File "/home/dellard/RF/QGL/QGL/Compiler.py", line 585, in propagate_node_frame_to_edges
    edge = ChannelLibraries.channelLib.connectivityG.edges[predecessor, chan]['channel']
TypeError: 'method' object is not subscriptable
```

dieris, 2017-11-15:
Do you have networkx 2.0 installed? See https://github.com/BBN-Q/QGL/pull/130/commits/4d851f527c0cab1533444b1b34297c4763ba75ef

dellard, 2017-11-15:
No: looks like I have 1.11.  Doing a conda update now.

dellard, 2017-11-15:
That solved the problem.  Maybe we should have a ticket for testing package dependencies?

closed 2017-11-15 by dellard