BBN-Q/QGL #150: the compiler modifies the input sequences

dellard, 2018-01-26:
`compile_to_hardware` modifies the sequence lists it is passed.  In the past, this was mostly harmless
because nobody looked at the sequence lists after the compiler was done with them.
The method for compiling the TDM instructions, however, needs the original sequences.

The workaround is to `copy` the sequence lists (a shallow copy seems sufficient) before passing
them to the compiler, but the compiler could do this itself _unless_ there's some reason that we
actually want the compilation process to be destructive.

dieris, 2018-07-25:
This question is still relevant, but tdm instructions are now generated from the compiled sequence: https://github.com/BBN-Q/QGL/blob/446ff4f86b5ae9938a2c9a7953ec1b7adefdd84f/QGL/Compiler.py#L444

matthewware, 2018-07-26:
I would still vote to copy them.  It's causing issues with the axis descriptor creation.  In that case I found a shallow copy wasn't sufficient.