BBN-Q/QGL #155: 'drag' shapeFun doesn't work on q2

jaseung, 2018-02-16:
This is really weird, but I would say that 'drag' shapeFun is not applied to q2(qubit) even though 'drag' is selected in ExpSettingGUI and ChannelParams.json file looks right. I have both q1 and q2 set up with drag. When a qubit pulse is plotted in jupyter notebook, I see  only q1 with DRAG pulse, not q2. q2 shows only gaussian waveform. Also, in real measurement, I verified that the DRAGCal does not work, indication there's really no DRAG on q2.  I even tried adding q3 and q3 doesn't work with DRAG, either.

I am currently using this branch for QGL : https://github.com/Plourde-Research-Lab/QGL/tree/JPM. Also, I attached a zip file that contains json files (ChannelParams, Instruments, Measurements, Sweeps, and DefaultExpSettings).

[DRAGonq2NotWork.zip](https://github.com/BBN-Q/QGL/files/1732196/DRAGonq2NotWork.zip)

dieris, 2018-02-16:
The amplitude of your drag correction is set by the `dragScaling` parameter. It is set to 0 in your ChannelParams, so I think that's why you're seeing a simple Gaussian.

jaseung, 2018-02-16:
You are right; the dragScaling was set to zero in the json file attached. However, even if I put non-zero dragScaling, the plot for q2 in jupyter notebook looks the same---gaussian.  As I mentioned, the DRAGCAL in qlab during which the dragScaling is swept just fails with q2. I know there is nothing special about q2, but only q1 works with drag. :( Because the plot is already wrong, this looks like something to do with QGL.

closed 2018-04-12 by jaseung

dieris, 2018-04-12:
@jaseung were you able to find where the problem was? Sorry for letting this slip, but it's not always easy for us to debug on a different fork

jaseung, 2018-04-12:
Hello Diego,

Actually, I don't know where the problem was. I had a chance to try DRAG
calibration on both q1 and q2, and they worked fine. I would just think for
now that some settings were not correct or who knows what. Next time I
encounter the same problem again, then I plan to post it again. Thanks for
asking.

Jaseung