BBN-Q/QGL #162: Two qubit RB sequence problem

jaseung, 2018-04-25:
I found that two-qubit RB sequence created by QGL is not right. Shown below is the 4 Clifford gates for two qubit RB, measured with our oscilloscope. The pulse sequence plot in the jupyter notebook looks the same as in the oscilloscrope, so I thought this would be a QGL issue.

The problem is that the qubit pulse train appears to go out of sync with the gate pulses so a gap appears between the ends of gate and qubit pulse train. 

Can you see if this is reproducible in your QGL and if yes, would you fix it?

[image]

dieris, 2018-04-27:
@jaseung please try https://github.com/BBN-Q/QGL/pull/163

dieris, 2018-04-30:
I actually can't reproduce this issue anymore after the fix above. Can you try to set all delays to 0 and see where your markers go out of sync? Is it at a specific point in the sequence, or is it cumulative?

jaseung, 2018-04-30:
Hmm, it's the same. I set all delays to 0 and still see the same problems (gate and qubit out of sync). I even tried with markers disabled, but no luck. To see if there is any difference, I also tried to use QGL-last-matlab release, but the result is the same. Which QGL version did you use to test the fix? Any suggestion?

Just in case you're interested, I attached the ChannelParams.json below.

*ChannelParams.zip
[ChannelParams.zip](https://github.com/BBN-Q/QGL/files/1961620/ChannelParams.zip)

*Screencapture of pulse trains in jupyter-notebook.
[image]

dieris, 2018-04-30:
I'm only testing on the master branch. Unfortunately we do not support the matlab-compatible version anymore. It seems that your settings are alright, though. What happens when you disable the `gatechan` for both `q2` and `CR12`?

jaseung, 2018-04-30:
With gate channels for q2 and CR12, there's still the same issue. Well, I
guess I should try the master branch.  Can you provide a measure.yml for me
to use in QGL to create two-qubit RB?

Jaseung

jaseung, 2018-05-01:
Sorry, I meant ' With gate channels for q2 and CR12 disabled'.

Jaseung

dieris, 2018-05-01:
Sent by email

dieris, 2018-05-25:
Remaining issue, reported by @jaseung : control qubit gate (marker) stays high when combined with a gate on the CR pulse in the same channel

closed 2019-04-12 by jaseung