BBN-Q/QGL #172: `compile_to_hardware` mangles sequences

matthewware, 2018-07-25:
Is there any danger in doing a deep copy of seqs before we get here:

https://github.com/BBN-Q/QGL/blob/446ff4f86b5ae9938a2c9a7953ec1b7adefdd84f/QGL/Compiler.py#L327-L331

This mangles the seqs and even makes lines like 
```python
tomo_block = state_tomo([X(q1)], (q1,))
cals = create_cal_seqs((q1,), numRepeats=4)
seqs = tomo_block + cals

axis_descriptor = [{
    'name': 'segment num',
    'unit': None,
    'points': list(range(0,len(tomo_block))),
    'partition': 1
}]
axis_descriptor.append(cal_descriptor((q1,), 4))

metafile = compile_to_hardware(seqs, 'Tomo/Tomo', axis_descriptor=axis_descriptor, 
                               extra_meta = {'sequences':tomo_block})
```
crash hard since the metadata construction is happening *after* this initial loop.  This seems to work for me with minor changes to QGL's `Pulse` object:
```python
compiler_seqs = copy.deepcopy(seqs)
metafile = compile_to_hardware(compiler_seqs, 'Tomo/Tomo', axis_descriptor=axis_descriptor, 
                               extra_meta = {'sequences':str(tomo_block)})
```

Assuming we're not doing something like GST (a case we can work around) is there any reason not to do this?  Seems this would be an issue anytime you needed seqs or adata about them after you'd run `compile_to_hardware`

matthewware, 2018-07-25:
I basically just added a __copy__ and __deepcopy__ method

dieris, 2018-07-25:
I think we can indeed copy the sequence (but do we need to define copy methods for lower layers? Note that tests are not passing now). This question was also brought up in https://github.com/BBN-Q/QGL/issues/150

Note however that for this application, the sequence is already saved in `Tomo-code.json` before the `WAIT` is added https://github.com/BBN-Q/QGL/blob/446ff4f86b5ae9938a2c9a7953ec1b7adefdd84f/QGL/Compiler.py#L325

matthewware, 2018-07-26:
In my experience you need to have a `__deepcopy__` method defined for all the objects to be deep copied.  Here Pulse is the base object.  The code above doesn't work without a __deepcopy__ method.  I guess I'm fine with not copying and loading the sequences from json as long as we document that process and put up some examples.  My one concern is it starts to make creating axis descriptors comically obtuse.  @grahamrow I think it's the extra meta data Auspex needs down the line that causes issues here.  Do you have strong opinions on save-v-load?