BBN-Q/QGL #195: Marker comes after measurement if they are the same length

grahamrow, 2019-02-27:
```python
q1.measure_chan.pulse_params['length'] = 1000e-9
q1.measure_chan.trig_chan.pulse_params['length'] = 100e-9
plot_pulse_files(RabiAmp(cl["q1"], np.linspace(-1, 1, 11)), time=True)
```
[image]
```python
q1.measure_chan.pulse_params['length'] = 100e-9
```
Here we see the issue: instead of being simultaneous, the marker is AFTER the measure pulse. A whole bunch of the QGL test waveforms fall into this scenario. Could be a plotting issue, need to investigate further.
[image]
```python
q1.measure_chan.trig_chan.pulse_params['length'] = 150e-9
```
[image]

dieris, 2019-02-27:
I think it's real, but I also thought I had fixed it here: https://github.com/BBN-Q/QGL/pull/189

grahamrow, 2019-02-28:
Indeed. Cherry-picked fix from #189. Closing.

closed 2019-02-28 by grahamrow