BBN-Q/QGL #231: ChannelLibrary database notes not set

matthewware, 2019-09-24:
This might be a `bbndb` issue as well.  Seems like the notes attribute of the data base isn't getting set.  Here's a minimal working example:

```python
from QGL import *                                                                                                                                                      

cl = ChannelLibrary()cl = ChannelLibrary(db_resource_name=":memory:") 
q1 = cl.new_qubit("q1") 
cl.ls()

AttributeError     Traceback (most recent call last)
__> 1 cl.ls()

~/Github/QGL/QGL/ChannelLibraries.py in ls(self)
    139     def ls(self):
    140         cdb = Channels.ChannelDatabase
__> 141         q = self.session.query(cdb.label, cdb.time, cdb.id, cdb.notes).\
    142             order_by(Channels.ChannelDatabase.id, Channels.ChannelDatabase.label, Channels.ChannelDatabase.notes).all()
    143         table_code = ""

AttributeError: type object 'ChannelDatabase' has no attribute 'notes'

```

The database *does* have a notes attribute but it seems to not be initialized.

I did a pip install of develop on QGL, Auspex and master of bbndb.

matthewware, 2019-09-24:
@grahamrow

matthewware, 2019-09-24:
This issued also breaks The Example-Channel-Lib notebook (Example-Q2).

dieris, 2019-09-25:
It works for me when creating a new database from `:memory:`. Have you tried restarting the notebook?
It will fail with `example.sqlite` because it was created without the `notes` attribute. I'll push an updated version

matthewware, 2019-09-25:
hmm I couldn't get `:memory:` to work for me yesterday.  Let me try on another computer.

matthewware, 2019-09-25:
You're right...  A fresh install on a different computer works...  Okay I'll try to figure this out.  Closing for now.

closed 2019-09-25 by matthewware