BBN-Q/QGL #240: Store TDM word size and memory size in APSPattern

ahelsing, 2019-10-17:
Store the TDM Word size and the size of the TDM memory / number of words / valid address range (pick something) in a constant in `APS*Pattern.py`
* QGL2 may use this to manage memory for programmers

sfallek1, 2019-10-31:
@ahelsing the word size is 32 bits and the address range is 0->4095 (see tdm hdl source Customize.vhd). I'm not sure what branch you are on, but this can go in the header of the pattern gen whenever you need it. I doubt we will run out of addresses in the medium to long term, but we can even expand this if need be

ahelsing, 2019-11-01:
I work off develop generally. Care to make the change to add it there?

sfallek1, 2019-11-01:
sounds good, see d90a3333 MAX_VRAM_ADDRESS

closed 2019-11-01 by sfallek1