BBN-Q/QGL #249: QGL examples have database related errors

rblazarus, 2020-01-16:
Using branch develop, QGL examples have database related errors. Exhibits in Ex2 and Ex3. 

Creating engine...

---------------------------------------------------------------------------
OperationalError                          Traceback (most recent call last)
//anaconda3/lib/python3.7/site-packages/sqlalchemy/engine/base.py in _execute_context(self, dialect, constructor, statement, parameters, *args)
   1243                     self.dialect.do_execute(
-> 1244                         cursor, statement, parameters, context
   1245                     )

//anaconda3/lib/python3.7/site-packages/sqlalchemy/engine/default.py in do_execute(self, cursor, statement, parameters, context)
    549     def do_execute(self, cursor, statement, parameters, context=None):
--> 550         cursor.execute(statement, parameters)
    551 

OperationalError: no such column: receiver.reference_freq

The above exception was the direct cause of the following exception:

OperationalError                          Traceback (most recent call last)
<ipython-input-2-17d69b72b113> in <module>
----> 1 cl = ChannelLibrary(db_resource_name="./example.sqlite")

~/Repos/QGL/QGL/ChannelLibraries.py in __init__(self, db_resource_name, db_provider)
    113             self.session.commit()
    114 
--> 115         self.update_channelDict()
    116 
    117         # Update the global reference

dieris, 2020-01-16:
Should be fixed by https://github.com/BBN-Q/QGL/commit/b8c7620bbe2d8719b65c13e99ba3660a7adb0bf4

dieris, 2020-01-27:
example.sqlite has been removed

closed 2020-01-27 by dieris